paulbovbel/frontier_exploration #6: Footprint filter

paulbovbel, 2015-02-23:
Need to filter out goals that fall inside robot footprint, behaves poorly with some local planners.

kushaagra, 2015-06-11:
Yup, this needs to be done. Various times the local planner fails to produce a path to the frontier given and gives an error, leading to shutting down of the frontier exploration.

Patagonicus, 2015-07-03:
We, a small group of students, are also affected by this bug.

We're willing to look into the code to see if we're able to fix this, but we don't have a lot of experience with coding for ROS - we're mostly just making different nodes work together for our project and frontier_exploration is one of the last few that sometimes fail to work as intended. Any pointers for where to look/how to implement this would be great.

paulbovbel, 2015-07-03:
@Patagonicus, I would certainly appreciate any help you'd be willing to give. While I'm hoping to put some time aside for a thorough refactor, this behaviour could be fixed at the source:

https://github.com/paulbovbel/frontier_exploration/blob/indigo-devel/src/frontier_search.cpp#L146

This function should return false for any cells caught within the robot footprint.

paulbovbel, 2015-07-03:
To that end, you may find the following functions useful:

http://docs.ros.org/jade/api/costmap_2d/html/classcostmap__2d_1_1Costmap2DROS.html#a8e23eef773096a1fca61ae038220070d

http://docs.ros.org/indigo/api/frontier_exploration/html/namespacefrontier__exploration.html#a9b90012ac183da448a95eec412936066