paulbovbel/frontier_exploration #11: Failed Exploration

kushaagra, 2015-06-11:
I get failed exploration problem a lot of times. Sometimes it works pretty well and sometimes it gives failed exploration problems. Is there  an issue with my usage?

paulbovbel, 2015-09-01:
I would need a lot more information to find out.

kushaagra, 2015-10-21:
Well, I solved the problem.
There is an issue with your code. You have not maintained a list of inaccessible frontiers. Due to this your code fails a lot. But after I changed your code to include that feature,it works perfectly fine.

paulbovbel, 2015-10-21:
It's doing a breadth first search from the robot position, which I thought would cover that. Can you make a PR?

sunnywinter, 2016-04-24:
hi I meet the same problem too,but I can't figure out.Can U teach me how to fix code to solve exploration problems

kushaagra, 2016-04-24:
I don't remember exactly how I did it. Neither do I have the code now. (I did it as an intern in a company, the company has the code). But basically as far as I remember,  each time a failed exploration came, the main reason was that the frontier was unreachable via the navigation library.
I did the following changes (not exactly but somewhat):
1.) For each frontier in the frontier list, change the location of the frontier to the nearest clear cell. (kind of ad-hoc method, but works in practice)
2.) Maintain an additional Inaccessible frontier list. Each time a frontier is failed to reach, add the frontier to that list. So next time when you generate a new frontier list, remove those frontiers which are in Inaccessssible frontier list. 
3.) After doing 2, you will notice that, some frontiers which are inaccessbile, can become accessible sometime later while exploring. So i set a fixed size for inaccessible list like 2 or 3, by hand tuning basically, so that for my application it worked perfectly. 

I hope you understood what I did.
I am really sorry that I can't provide the changed code. :(, because I was just an intern and not a permanent employee, so I do not have access to the code now.

AravindaDP, 2016-12-07:
To add my observations on this,

I have lot of failed explorations because the goal passed to move_base sometimes falls really close to an obstacle. And unfortunately it seems move_base currently has a bug that prevent it from considering planning to a point that is reachable within a search radius.
https://github.com/ros-planning/navigation/issues/262

However it seems we can specify a search radius in make_plan service exposed by the move_base. Probably we can use that service to check the reachability of the goal and potentially get a reachable point within a search radius that we can then pass to move_base as the goal?

paulbovbel, 2016-12-08:
That's unpleasant. Other than the change you propose (in navigation stack), you can crank up the inflation radius on the exploration costmap to prevent the planner from trying to explore too close to obstacles.

closed 2017-06-11 by paulbovbel