paulbovbel/frontier_exploration #12: Failed reaching frontiers

fuesschen, 2015-09-01:
I have a strange problem, my robot starts to explore, finds frontiers and starts to navigate to the frontiers in the correct order starting with the nearest. At some points the robot navigates to the point which is marked as frontier, reaches the point but it looks like the callback does not recognize the point as reached. The robot stands still but driving forward manually and let the robot navigate back to the frontier mostly solves the problem.
May this failure belongs to the move_base node or to the frontier exploration node?

paulbovbel, 2015-09-01:
I would guess it's something with move_base, though maybe some retry/timeout logic in frontier_exploration wouldn't hurt either. Thanks for letting me know.

fuesschen, 2015-09-07:
The xy and yaw goal tolerances were also my first thought, so i already tried out to alter the values. Looks like i have to tweak them more. 
Thanks for your help, i will leave a comment if i find the failure elsewhere.

closed 2015-09-07 by fuesschen