paulbovbel/frontier_exploration #17: cannot launch node of type [frontier_exploration/explore_client]: can't locate node [explore_client] in package [frontier_exploration] ERROR: cannot launch node of type [frontier_exploration/explore_server]: can't locate node [explore_server] in package [frontier_exploration]

NaqiNick, 2016-01-02:
i have solved the previous issue but now it is giving the following error
 cannot launch node of type [frontier_exploration/explore_client]: can't locate node [explore_client] in package [frontier_exploration]
ERROR: cannot launch node of type [frontier_exploration/explore_server]: can't locate node [explore_server] in package [frontier_exploration]

paulbovbel, 2016-05-09:
Sorry, but this looks like an issue with how you setup the workspace, and not an issue with the code/package. Read the catkin workspace tutorials and ask on ROS answers if you're still having trouble.

closed 2016-05-09 by paulbovbel

NaqiNick, 2016-05-09:
thank you for your reply yes that issue was with my configuration i reinstalled ROS and tried to launch the no_global.launch file and now it seems to launch properly but after i select the polygon boundaries and the center point from the RVIZ the program seems to halt at the if statements inside the explore_server.cpp file which is if(!move_client_.waitForServer() || !updateBoundaryPolygon.waitForExistence() || !getNextFrontier.waitForExistence()) it seems that the simpleClients are waiting for their server to come online but the server never comes online i think and nothing happens it just seems to stop at this statement and doesn't move on i seriously need your help here please do reply.

Leiqtcv, 2019-12-10:
hello, because you don't have package frontier_exploration installed in your work space. see the description: http://wiki.ros.org/frontier_exploration/
you can tip "rospack find frontier_exploration", i think it shows not found. then just use sudo apt-get install. hope can eventually help.

YXGuan, 2022-04-16:
> hello, because you don't have package frontier_exploration installed in your work space. see the description: http://wiki.ros.org/frontier_exploration/ you can tip "rospack find frontier_exploration", i think it shows not found. then just use sudo apt-get install. hope can eventually help.

I am having the same issue even though frontier_exploration has been installed
[image]

Your help would be much appreciated.

QingquanBao, 2022-11-29:
> I am having the same issue even though frontier_exploration has been installed [image]
>
> Your help would be much appreciated.

Hi, did you solve this problem? I am following this [husky_guide](http://www.clearpathrobotics.com/assets/guides/melodic/husky/HuskyFrontiers.html) and failed to run, even if I `catkin_make clean` and remake the package.

I would appreciate any advice!!!!