paulbovbel/frontier_exploration #19: How do you use actionlib to send a bounded exploration goal?

pgigioli, 2016-04-04:
I understand how to use actionlib to send an unbounded exploration goal (simply send a goal with a blank polygon) but I want to send a bounded exploration goal using actionlib.  

My code is set up like this:

`frontier_exploration::ExploreTaskGoal goal;

  geometry_msgs::PolygonStamped boundary;
  geometry_msgs::Point32 p1, p2, p3, p4, p5;

  p1.x = dim/2; p1.y = dim/2; p1.z = 0.0;
  p2.x = dim/2; p2.y = -dim/2; p2.z = 0.0;
  p3.x = -dim/2; p3.y = -dim/2; p3.z = 0.0;
  p4.x = -dim/2; p4.y = dim/2; p4.z = 0.0;
  p5.x = dim/2; p5.y = dim/2; p5.z = 0.0;

  boundary.polygon.points.reserve(5);
  boundary.header.seq = 0;
  boundary.header.stamp = ros::Time::now();
  boundary.header.frame_id = "map";
  boundary.polygon.points[0] = p1;
  boundary.polygon.points[1] = p2;
  boundary.polygon.points[2] = p3;
  boundary.polygon.points[3] = p4;
  boundary.polygon.points[4] = p5;

  geometry_msgs::PointStamped start_point;
  start_point.point.x = 0.0; start_point.point.y = 0.0; start_point.point.z = 0.0;

  goal.explore_center = start_point;
  goal.explore_boundary = boundary;
  client.sendGoal(goal)`

The problem is that the boundary never gets sent and I end up with unbounded exploration no matter how small I make the boundary dimensions.  The output of "cout << boundary << endl;" shows that the polygon.points[] is blank.  Am I initiating the boundary polygon correctly?  Thanks!

paulbovbel, 2016-04-04:
try boundary.polygon.points.push_back[p1], etc.

You can always rostopic echo the goal topic of the action server to check what the submitted goal message from your client contains.

pgigioli, 2016-04-04:
Using push_back worked!  Thanks so much for your timely response.

Paul Gigioli

-- 
*Paul M Gigioli*

PwC | Financial Services Advisory
Office: Chicago | Mobile: +1 (872) 202-1845
Email: paul.m.gigioli@pwc.com
PricewaterhouseCoopers LLP
One North Wacker, Chicago, Illinois, 60606, United States of America
http://www.pwc.com/us

______________________________________________________________________
The information transmitted, including any attachments, is intended only for the person or entity to which it is addressed and may contain confidential and/or privileged material. Any review, retransmission, dissemination or other use of, or taking of any action in reliance upon, this information by persons or entities other than the intended recipient is prohibited, and all liability arising therefrom is disclaimed. If you received this in error, please contact the sender and delete the material from any computer. PricewaterhouseCoopers LLP is a Delaware limited liability partnership.  This communication may come from PricewaterhouseCoopers LLP or one of its subsidiaries.

closed 2016-04-04 by pgigioli

msapack, 2018-04-04:
How do you send a goal with a blank polygon?