paulbovbel/frontier_exploration #21: Illegal bounds change, was [tl: ...., but is now ...]. The offending layer is explore_costmap/explore_boundary

jamchen1025, 2016-08-18:
I have a problem when using frontier_exploration, move_base, rtabmap for unknown map exploration.

Below are the log messages in layered_costmap.cpp: 
(100.... -> 1e30, -100... ->-1e30)
=================================================
global_costmap/static_layer
    prev [tl: (100......., 100.......), br: (-100....., -100.....)]
    now  [tl: (-25.475000, -25.475000), br: (25.525001, 25.525001)]
global_costmap/obstacle_layer
    prev [tl: (-25.475000, -25.475000), br: (25.525001, 25.525001)]
    now  [tl: (-25.475000, -25.475000), br: (25.525001, 25.525001)]
global_costmap/inflation_layer
    prev [tl: (-25.475000, -25.475000), br: (25.525001, 25.525001)]
    now  [tl: (-25.975000, -25.975000), br: (26.025001, 26.025001)]
---
local_costmap/obstacle_layer
    prev [tl: (100......., 100.......), br: (-100....., -100.....)]
    now  [tl: (-1.450000, -1.600000), br: (0.197008, 0.062753)]
local_costmap/inflation_layer
    prev [tl: (-1.450000, -1.600000), br: (0.197008, 0.062753)]
    now  [tl: (-1.950000, -2.100000), br: (0.715967, 0.579622)]

explore_costmap/static_layer
    prev [tl: (100......., 100.......), br: (-100....., -100.....)]
    now  [tl: (-25.475000, -25.475000), br: (25.525001, 25.525001)]
---
explore_costmap/explore_boundary
    prev [tl: (-25.475000, -25.475000), br: (25.525001, 25.525001)]
    now  [tl: (-25.500000, -25.500000), br: (25.475001, 25.475001)]
    >>> show Illegal bounds change WARN

explore_costmap/obstacle_layer
    prev [tl: (-25.500000, -25.500000), br: (25.475001, 25.475001)]
    now  [tl: (-25.500000, -25.500000), br: (25.475001, 25.475001)]
explore_costmap/inflation_layer
    prev [tl: (-25.500000, -25.500000), br: (25.475001, 25.475001)]
    now  [tl: (-26.000000, -26.000000), br: (25.975001, 25.975001)]
=================================================

The warning message is because "now" br is small than "prev" br.
Is there possible initial setting wrong and cause this warning?

paulbovbel, 2016-08-25:
Possibly, I'm not sure. Can you provide a reproduction case with a gazebo environment?

paulbovbel, 2016-08-25:
Looks like inflation increases the br size as well

jamchen1025, 2016-08-26:
Hi, @paulbovbel , 
Sorry  I am a newbie to ROS,
Don't understand what do I need to provide exactly,
Do U mean gazebo word file(sdf)? or parameter setting files(costmap, move_base...).
Thanks!

paulbovbel, 2016-08-27:
A set of files that I could `roslaunch ...` to reproduce the problem would be fantastic, so all of the above if possible.

jamchen1025, 2016-08-29:
Hi, @paulbovbel ,
Can U give me your email address, so I can email u related files privately, thanks!

paulbovbel, 2016-09-06:
paul@bovbel.com

closed 2017-06-11 by paulbovbel

allanis79, 2017-06-27:
Hello @paulbovbel ,
I am getting the same error

`[ WARN] [1498586157.959931982, 1836.780000000]: Illegal bounds change, was [tl: (-99.975000, -99.975000), br: (100.025003, 100.025003)], but is now [tl: (-100.000000, -100.000000), br: (99.975003, 99.975003)]. The offending layer is explore_costmap/explore_boundary`

Can you help me to solve this? I tried figuring out why it is happening, but i was not able to solve it.

Thanks.

paulsammut, 2017-12-08:
Me too! 

[ WARN] [1512753970.282270573, 1691.613000000]: Illegal bounds change, was [tl: (-5.500000, -5.500000), br: (5.500000, 5.500000)], but is now [tl: (-5.525000, -5.525000), br: (5.450000, 5.450000)]. The offending layer is explore_costmap/explore_boundary

I'm using rtabmap posting up the map on /grid_map (which i remapped to map). Makes to different if I have resize_to_boundary true or false.