paulbovbel/frontier_exploration #28: clarification ~explore_clear_space (bool, default: true)

JoBot88, 2017-06-21:
Good morning and thank you in advance for your help,

My problem is the following, I have a robot that navigates in an area potentially hosting obstacles made of glass that my robot could not see with its laser. On my robot I have bumpers and sonars that could sense these glasses. Is there a simple way for me to make this package work by using my other sensors data in order to prevent collisions but still explore everything that is explorable ?
 
Thank you

paulbovbel, 2017-07-07:
You can make use of additional sensor data by integrating custom costmap_2d layer plugins.

The parameter explore_clear_space is documented here: http://wiki.ros.org/frontier_exploration#Parameters-1

closed 2017-07-07 by paulbovbel

lukelu0520, 2020-09-29:
Dear frontier_exploration friends,

Creating extra custom costmap_2d layer plugins is cool but doesn't sound a unique and simplest way to solve the problem.
Since you've created a nice global_map.launch file, could we simply extend ObstacleLayer plugin, by adding additional observation sources ? such as:
> observation_sources: laser bumper sonar
> laser: {data_type: LaserScan, ...} 
> bumper: {data_type: BumperEvent, ...} 
> sonar: {data_type: ...}

Hope to hear your expertise.

-- Luke