paulbovbel/frontier_exploration #29: Illegal bounds change

allanis79, 2017-06-27:
Hello @paulbovbel ,

I tried follwing the documentation in ROS WIKI to run froniter exploration, but when i try running it I am getting the follwoing error: 

`[ WARN] [1498586157.959931982, 1836.780000000]: Illegal bounds change, was [tl: (-99.975000, -99.975000), br: (100.025003, 100.025003)], but is now [tl: (-100.000000, -100.000000), br: (99.975003, 99.975003)]. The offending layer is explore_costmap/explore_boundary`

I tired to debug it, but I was not able to do it. Can you help me sovle this?

Thanks.

alefagg, 2017-06-30:
Hello, 
have you solved this problem? I am fighting with the same type of warning.

Thanks

allanis79, 2017-07-06:
@alefagg 

No I didnot. How about you? I tried really hard to findout what's happenbing, but i was not able to debug why it is happening.

Please let me know if you solved it.

paulbovbel, 2017-07-07:
Thanks for the report. It looks like there may have been a change upstream in the costmap_2d package that's issuing this warning. I'll try to have a look at this sometime in the near future, but PRs are welcome.

jaredjohansen, 2019-01-24:
I am getting the same warning.  I will try and add some more information in the hopes that @paulbovbel (or anyone else) can help me find the root cause of this (since it appears many people have run into this issue...but I have not seen any solutions posted).

I have created a new catkin workspace and dumped the latest frontier_exploration, navigation, and geometry packages into it.  (Any other dependencies are system installs, which would be a bit older.)  I build this new workspace without any issue.

I take the provided exploration.launch file and modify to fit my system.  It looks like this:
```
<launch>
   <arg name="plugin" default="exploration_server::ExamplePlugin"/>
   <param name="plugin_name" value="$(arg plugin)"/>
   <arg name="blacklist_box_size" default="0.5"/>
   <param name="blacklist_box_size" value="$(arg blacklist_box_size)"/>

   <!-- Set to your sensor's range -->
   <arg name="sensor_range" default="6.0"/>

   <node pkg="exploration_server" type="plugin_client" name="plugin_client" output="screen">
   </node>

   <node pkg="exploration_server" type="exploration_server_node" name="exploration_server_node" output="screen">
       <rosparam param="points" ns="explore_costmap" file="$(find exploration_server)/config/points.yaml" command="load"/>
       <param name="frequency" type="double" value="2.0"/>
       <param name="goal_aliasing" type="double" value="$(arg sensor_range)"/>

       #All standard costmap_2d parameters as in move_base, other than PolygonLayer
       <rosparam ns="explore_costmap" subst_value="true">
           footprint: [[-0.5, -0.33], [-0.5, 0.33], [0.65, 0.33], [0.65, -0.33]]
           footprint_padding: 0.02

           transform_tolerance: 0.5
           update_frequency: 4.0
           publish_frequency: 3.0

           #must match incoming static map
           global_frame: map
           robot_base_frame: base_link
           resolution: 0.05

           obstacle_range: 5.5
           raytrace_range: 6.0

           rolling_window: false
           track_unknown_space: true

           plugins:
               - {name: static,              type: "costmap_2d::StaticLayer"}
               - {name: polygon_layer,       type: "polygon_layer::PolygonLayer"}
               #Can disable sensor layer if gmapping is fast enough to update scans
               - {name: obstacles_laser,     type: "costmap_2d::ObstacleLayer"}
               - {name: inflation,           type: "costmap_2d::InflationLayer"}

           static:
               #Can pull data from gmapping, map_server or a non-rolling costmap
               map_topic: /map
               # map_topic: move_base/global_costmap/costmap
               subscribe_to_updates: true

           polygon_layer:
               resize_to_polygon: true

           obstacles_laser:
               observation_sources: laser
               laser: {data_type: LaserScan, clearing: true, marking: true, topic: scan, inf_is_valid: true}

           inflation:
               inflation_radius: 1.0
               cost_scaling_factor: 2.0

       </rosparam>
   </node>
</launch>
[... 12 lines cut ...]
[ INFO] [1548362226.059416182]: Resizing costmap to 323 X 460 at 0.050000 m/pix
[ INFO] [1548362226.159046920]: Received a 323 X 460 map at 0.050000 m/pix
[ INFO] [1548362226.159073839]: Subscribing to updates
[ INFO] [1548362226.165216388]: Using plugin "polygon_layer"
[ INFO] [1548362226.175201664]: Using plugin "obstacles_laser"
[ INFO] [1548362226.176616850]:     Subscribed to Topics: laser
[ INFO] [1548362226.191684321]: Using plugin "inflation"
[ WARN] [1548362239.458079619]: Please select an initial point for exploration inside the polygon
[ INFO] [1548362240.561564160]: Sending goal
[ INFO] [1548362240.569826937]: Updating polygon
[ INFO] [1548362240.569848362]: Requesting a goal
Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
        at line 108 in /tmp/binarydeb/ros-kinetic-class-loader-0.3.9/src/class_loader.cpp
[ WARN] [1548362240.713385327]: Illegal bounds change, was [tl: (1.513761, -6.792992), br: (17.663761, 16.207009)], but is now [tl: (1.488761, -6.817992), br: (8.149106, 2.933292)]. The offending layer is explore_costmap/polygon_layer
[ WARN] [1548362240.713435744]: The origin for the sensor at (0.59, -0.04) is out of map bounds. So, the costmap cannot raytrace for it.
[ INFO] [1548362241.523284138]: Resizing costmap to 323 X 460 at 0.050000 m/pix
[ WARN] [1548362242.713388221]: Illegal bounds change, was [tl: (-6.660397, -14.142630), br: (9.489603, 8.857370)], but is now [tl: (1.488761, -6.817992), br: (8.149106, 2.933292)]. The offending layer is explore_costmap/polygon_layer
[ WARN] [1548362243.713390684]: Illegal bounds change, was [tl: (-6.660397, -14.142630), br: (9.489603, 8.857370)], but is now [tl: (1.488761, -6.817992), br: (8.149106, 2.933292)]. The offending layer is explore_costmap/polygon_layer
```

The robot does not move.  I suspect it's due to the warnings above.  Any suggestions as to where I went awry (or how to fix this so that my robot explores properly) would be appreciated!

A few other (perhaps relevant) notes:
- I am running ROS kinetic.
- I am running on a Clearpath Husky (not simulation)

erinline, 2019-03-03:
I am seeing exactly this error in ROS melodic. I have tried running the frontier exploration demo on both the Husky and the turtlebot3 (both in gazebo) and get the same error with both. Has anyone figured out the fix?

jaredjohansen, 2019-03-06:
@erinline, I did not find a solution.  I ended up using another exploration package (explore_lite), which I was able to get up and running.  It works in a similar fashion.  It computes all the unexplored frontiers (although it does this for the whole map...rather than within a particular polygon) and explores them in a greedy fashion.

erinline, 2019-03-06:
@jaredjohansen thanks for the info!! I'll try that package

kuailefeifei, 2023-02-22:
I can provide a possible solution, 
polygon_layer.cpp void PolygonLayer::updateBounds() {
  *min_x = std::min(min_x_, *min_x);
  *min_y = std::min(min_y_, *min_y);
  *max_x = std::max(max_x_, *max_x);
  *max_y = std::max(max_y_, *max_y);
}
hope this is helpful. :)