paulbovbel/frontier_exploration #43: "No such file" error during catkin_make with ROS Melodic

erinline, 2019-03-01:
I've just cloned the Frontier-exploration package, and I get an error when running catkin_make:

`In file included from /home/erinline/Documents/testsim-mod/src/frontier_exploration/polygon_layer/src/polygon_layer.cpp:1:0:
/home/erinline/Documents/testsim-mod/src/frontier_exploration/polygon_layer/include/polygon_layer/polygon_layer.h:9:10: fatal error: exploration_msgs/SetPolygon.h: No such file or directory
 #include <exploration_msgs/SetPolygon.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
frontier_exploration/polygon_layer/CMakeFiles/polygon_layer.dir/build.make:62: recipe for target 'frontier_exploration/polygon_layer/CMakeFiles/polygon_layer.dir/src/polygon_layer.cpp.o' failed
make[2]: *** [frontier_exploration/polygon_layer/CMakeFiles/polygon_layer.dir/src/polygon_layer.cpp.o] Error 1
CMakeFiles/Makefile2:9226: recipe for target 'frontier_exploration/polygon_layer/CMakeFiles/polygon_layer.dir/all' failed
make[1]: *** [frontier_exploration/polygon_layer/CMakeFiles/polygon_layer.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 19%] Generating Python code from SRV exploration_msgs/SetPolygon
[ 19%] Generating Python msg __init__.py for exploration_msgs
[ 19%] Generating Python srv __init__.py for exploration_msgs
[ 19%] Built target exploration_msgs_generate_messages_py
In file included from /home/erinline/Documents/testsim-mod/src/frontier_exploration/exploration_server/src/exploration_server.cpp:1:0:
/home/erinline/Documents/testsim-mod/src/frontier_exploration/exploration_server/include/exploration_server/exploration_server.h:8:10: fatal error: exploration_msgs/ExploreAction.h: No such file or directory
 #include <exploration_msgs/ExploreAction.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/build.make:62: recipe for target 'frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/src/exploration_server.cpp.o' failed
make[2]: *** [frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/src/exploration_server.cpp.o] Error 1
CMakeFiles/Makefile2:9527: recipe for target 'frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/all' failed
make[1]: *** [frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/all] Error 2
[ 19%] Built target exploration_msgs_generate_messages_eus
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`

I don't see the file on github either. Any suggestions how to resolve?

erinline, 2019-03-02:
Resolved, though I don't know why it worked. I had run catkin_make multiple times, and the header file was indeed generated, I located it on my computer in the correct place, but catkin_make never worked. Then I went through changing all the calls from 'ExploreAction' to 'Explore', since the action file is called 'Explore.action'. Still catkin_make did not compile, with the same error as before. So I went and changed them all back. Then catkin_make compiled.
Shrug.

closed 2019-03-02 by erinline

paulbovbel, 2019-03-03:
Try using catkin-tools instead of catkin_make

RDaneelOlivav, 2019-10-14:
Having same issue.
The .h files generated for the messages are generated but not found. Thats why after the second catkin_make it compiles. Any ideas on how to fix this to allow correct compilation on the first catkin_make? Without using catkin-tools I mean.

raushankumar586, 2020-10-06:
Running catkin_make multiple times, helped me also.  it has created the file.