paulbovbel/frontier_exploration #46: Cartographer Unable To Find Frontier

Rafcin, 2019-04-29:
With the current version of cartographer, frontier exploration and explore_lite have the issue where they are unable to read the map correctly due to how the cells are created. 

Would it be possible to update this to work with the cartographer map?

c650, 2019-06-07:
Bump. I am having this issue right now.

paulbovbel, 2019-06-07:
I'm not really sure the context of the change, but contributions (in terms of a deeper investigation into the issue, and a potential fix) are very welcome!

c650, 2019-06-08:
https://github.com/googlecartographer/cartographer/issues/1498 could it be some issue like this?

Rafcin, 2019-06-10:
https://github.com/HaoQChen/map_server

c650, 2019-06-10:
Does that make it work?