paulbovbel/frontier_exploration #47: melodic failed to request goal

JuliusSustarevas, 2019-05-27:
Hi,
I'm running melodic, properly setup navigation stack. 
I'm trying to use the point-click tool from Rviz and I find an issue very similar to

PSotiriadis, 2019-10-24:
Hi,
I had the same problem as you and I found where the problem is and a solution for that.
To explain, I will post two graphs. The first will be a rosgraph using explore_lite and the
second a rosgraph using frontier_exploration. 

 **explore_lite:**
[image]

**frontier_exploration:**
[image]

As you can see, in the first screenshot there is a bidirectional connection between explore node and move_base (move_base/action_topics) but in the second screenshot there is no connection between exploration_server_node and move_base. And exactly that is the problem you mentioned.

So in code you need to change exploration_server/src/exploration_server.cpp. The original (problematic) code is:
[image]

As you see exploration_server_node communicates with **_p3_001/move_base_** node which is not exist. Now you need to change the node to **_/move_base_** in order to be able to use frontier_exploration.
[image]

After this change the rosgraph for frontier_exploration will be:
[image]

But now there is another problem. **If a polygon is set inside map (turquoise color), then everything is working properly**. But exploration_server_node dies if the defined exploration polygon is bigger than the map (created from SLAM). For example if you choose this polygon:
[image]
you will receive the following error and frontier exploration will stop:
[image]

That means that if you use gmapping, you can configure initial map (see gmapping xmin, xmax, ymin, ymax parameters) to be big enough to satisfy your desired  exploration polygon. But if you want to use frontier_exploration with cartographer there is a Problem, because using cartographer you can not define the initial size of  the map and at the beginning you receive a map with size relative to your laser range(map is as small as possible to reduce CPU load). But I think its better to create a new issue for this problem.

I hope I helped you

kristfrizh, 2020-08-11:
Hello
I'm still encountered error after following yours way.

Is there anything else that i can do?

I'm using also melodic with jackal simulation
thank you

PSotiriadis, 2020-08-11:
Hi 
Can you display the error and a screenshot of your rosgraph??? I think it is another error.
Thanks

kristfrizh, 2020-08-11:
Hii
thank you for your fast reply
here are the error and the rosgraph

paulbovbel, 2020-08-11:
Hi folks, I apologise, I haven't had the time to bring the melodic branch up to snuff after a sizeable refactor - and hence never cut a melodic releaser. PRs are welcome and appreciated!

kristfrizh, 2020-08-13:
thank you for the hardwork
We will wait for your work at melodic
btw I'm sorry cannot help you because I'm quite new with this system and
need learn so much thing right now