paulbovbel/frontier_exploration #49: STILL ERROR: "No such file" error during catkin_make with ROS Melodic

RDaneelOlivav, 2019-10-14:
This issue was considered closed: 

But its not working. The catkin_make triggers this error all the time. Any idea on how to fix the import error of those .h generated for the messages? Becuase I tested creating a dummy binary thta includes those messages inside the same package, and worked , so it seems its a dependency issue of the packages that use these messages.

> Scanning dependencies of target exploration_server
> [ 46%] Building CXX object frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/src/exploration_server.cpp.o
> In file included from /home/user/catkin_ws/src/frontier_exploration/polygon_layer/src/polygon_layer.cpp:1:0:
> /home/user/catkin_ws/src/frontier_exploration/polygon_layer/include/polygon_layer/polygon_layer.h:9:10: fatal error: exploration_msgs/SetPolygon.h: No such file or directory
>  #include <exploration_msgs/SetPolygon.h>
>           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> compilation terminated.
> frontier_exploration/polygon_layer/CMakeFiles/polygon_layer.dir/build.make:62: recipe for target 'frontier_exploration/polygon_layer/CMakeFiles/polygon_layer.dir/src/polygon_layer.cpp.o' failed
> make[2]: *** [frontier_exploration/polygon_layer/CMakeFiles/polygon_layer.dir/src/polygon_layer.cpp.o] Error 1
> CMakeFiles/Makefile2:2424: recipe for target 'frontier_exploration/polygon_layer/CMakeFiles/polygon_layer.dir/all' failed
> make[1]: *** [frontier_exploration/polygon_layer/CMakeFiles/polygon_layer.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> In file included from /home/user/catkin_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:1:0:
> /home/user/catkin_ws/src/frontier_exploration/exploration_server/include/exploration_server/exploration_server.h:8:10: fatal error: exploration_msgs/ExploreAction.h: No such file or directory
>  #include <exploration_msgs/ExploreAction.h>
>           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

adamkrawczyk, 2020-02-21:
Have the same error and found a quick solution to use `catkin_make_isolated`

adamkrawczyk, 2020-02-27:
okay, here is a better solution not forcing to use catkin_make_isolated

```
1. cd <ros_ws>
2. catkin_make --only-pkg-with-deps exploration_msgs
3. catkin_make -DCATKIN_WHITELIST_PACKAGES=""
```

then you can use normally catkin make
these forces to build first exploration_msgs project then the rest is compiled

ZuyuanZhu, 2021-01-06:
> catkin_make -DCATKIN_WHITELIST_PACKAGES=""

I got the following errors after trying your step 3 catkin_make -DCATKIN_WHITELIST_PACKAGES="". And I got the same error if I continue with catkin_make.

`[  3%] Building CXX object frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/src/exploration_server.cpp.o
/home/user/simulation_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp: In member function 'bool exploration_server::ExplorationServer::inBoundary()':
/home/user/simulation_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:144:35: error: no matching function for call to 'costmap_2d::Costmap2DROS::getRobotPose(geometry_msgs::PoseStamped&)'
   costmap_ros_->getRobotPose(pose1);
                                   ^
In file included from /home/user/simulation_ws/src/frontier_exploration/exploration_server/include/exploration_server/base_plugin.h:7:0,
                 from /home/user/simulation_ws/src/frontier_exploration/exploration_server/include/exploration_server/exploration_server.h:4,
                 from /home/user/simulation_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:1:
/opt/ros/kinetic/include/costmap_2d/costmap_2d_ros.h:123:8: note: candidate: bool costmap_2d::Costmap2DROS::getRobotPose(tf::Stamped<tf::Transform>&) const
   bool getRobotPose(tf::Stamped<tf::Pose>& global_pose) const;
        ^
/opt/ros/kinetic/include/costmap_2d/costmap_2d_ros.h:123:8: note:   no known conversion for argument 1 from 'geometry_msgs::PoseStamped {aka geometry_msgs::PoseStamped_<std::allocator<void> >}' to 'tf::Stamped<tf::Transform>&'
/home/user/simulation_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp: In member function 'void exploration_server::ExplorationServer::requestAndSendGoal()':
/home/user/simulation_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:209:42: error: no matching function for call to 'costmap_2d::Costmap2DROS::getRobotPose(geometry_msgs::PoseStamped&)'
   costmap_ros_->getRobotPose(current_pose);
                                          ^
In file included from /home/user/simulation_ws/src/frontier_exploration/exploration_server/include/exploration_server/base_plugin.h:7:0,
                 from /home/user/simulation_ws/src/frontier_exploration/exploration_server/include/exploration_server/exploration_server.h:4,
                 from /home/user/simulation_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:1:
/opt/ros/kinetic/include/costmap_2d/costmap_2d_ros.h:123:8: note: candidate: bool costmap_2d::Costmap2DROS::getRobotPose(tf::Stamped<tf::Transform>&) const
   bool getRobotPose(tf::Stamped<tf::Pose>& global_pose) const;
        ^
/opt/ros/kinetic/include/costmap_2d/costmap_2d_ros.h:123:8: note:   no known conversion for argument 1 from 'geometry_msgs::PoseStamped {aka geometry_msgs::PoseStamped_<std::allocator<void> >}' to 'tf::Stamped<tf::Transform>&'
In file included from /usr/include/boost/smart_ptr/make_shared.hpp:15:0,
                 from /usr/include/boost/make_shared.hpp:15,
                 from /opt/ros/kinetic/include/ros/forwards.h:38,
                 from /opt/ros/kinetic/include/ros/common.h:37,
                 from /opt/ros/kinetic/include/ros/ros.h:43,
                 from /opt/ros/kinetic/include/tf2_ros/buffer.h:38,
                 from /opt/ros/kinetic/include/tf/tf.h:48,
                 from /opt/ros/kinetic/include/costmap_2d/layer.h:43,
                 from /opt/ros/kinetic/include/costmap_2d/layered_costmap.h:42,
                 from /opt/ros/kinetic/include/costmap_2d/costmap_2d_ros.h:41,
                 from /home/user/simulation_ws/src/frontier_exploration/exploration_server/include/exploration_server/base_plugin.h:7,
                 from /home/user/simulation_ws/src/frontier_exploration/exploration_server/include/exploration_server/exploration_server.h:4,
                 from /home/user/simulation_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:1:
/usr/include/boost/smart_ptr/make_shared_object.hpp: In instantiation of 'typename boost::detail::sp_if_not_array<T>::type boost::make_shared(Args&& ...) [with T = costmap_2d::Costmap2DROS; Args = {const char (&)[16], tf2_ros::Buffer&}; typename boost::detail::sp_if_not_array<T>::type = boost::shared_ptr<costmap_2d::Costmap2DROS>]':
/home/user/simulation_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:33:93:   required from here
/usr/include/boost/smart_ptr/make_shared_object.hpp:254:5: error: no matching function for call to 'costmap_2d::Costmap2DROS::Costmap2DROS(const char [16], tf2_ros::Buffer&)'
     ::new( pv ) T( boost::detail::sp_forward<Args>( args )... );
     ^
In file included from /home/user/simulation_ws/src/frontier_exploration/exploration_server/include/exploration_server/base_plugin.h:7:0,
                 from /home/user/simulation_ws/src/frontier_exploration/exploration_server/include/exploration_server/exploration_server.h:4,
                 from /home/user/simulation_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:1:
/opt/ros/kinetic/include/costmap_2d/costmap_2d_ros.h:80:3: note: candidate: costmap_2d::Costmap2DROS::Costmap2DROS(std::__cxx11::string, tf::TransformListener&)
   Costmap2DROS(std::string name, tf::TransformListener& tf);
   ^
[... cut ...]
frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/build.make:62: recipe for target 'frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/src/exploration_server.cpp.o' failed
make[2]: *** [frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/src/exploration_server.cpp.o] Error 1
CMakeFiles/Makefile2:5439: recipe for target 'frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/all' failed
make[1]: *** [frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j36 -l36" failed`