paulbovbel/frontier_exploration #53: Any plan for Melodic update?

ROBOTIS-Will, 2020-06-04:
Hi,
There was an inquiry regarding ROS 1 melodic support. Do you have any plan for this?
https://github.com/ROBOTIS-GIT/turtlebot3/issues/596
Thank you.

paulbovbel, 2020-08-11:
I don't have any immediate plans - PRs to integrate the melodic-devel frontier_exploration with turtlebot3, or the existing demo in https://github.com/husky/husky/blob/melodic-devel/husky_navigation/launch/exploration_demo.launch are welcome!

Dkaka, 2020-11-27:
This might be helpful? https://github.com/husky/husky/pull/150

civerachb-cpr, 2020-12-21:
If frontier_exploration is released & bloomed for melodic we can update husky_navigation accordingly and re-enable it.

Gavriel-CTO, 2022-11-14:
I pretty much have things running in Noetic (Package built from source) but have these two issues:

The initial exploration seems to be working but ends early at some point - depends on pre-determined points AFAIK... and I get this:

[ INFO] [1668333655.747359764, 1420.770000000]: Goal reached
[ERROR] [1668333655.747967449, 1420.770000000]: BUG: Got a second transition to DONE

Then if I trying a new polygon:

[ INFO] [1668417507.443129845, 140.190000000]: Sending goal
[ INFO] [1668417507.475213758, 140.210000000]: Updating polygon
[ INFO] [1668417507.475353658, 140.210000000]: Requesting a goal
Warning: class_loader.ClassLoader: SEVERE WARNING!!!
Attempting to unload (MY_PATH)/husky_ws/devel/lib//libexample_plugin.so
while objects created by this library still exist in the heap!
You should delete your objects before destroying the ClassLoader. The library will NOT be unloaded.
at line 119 in /tmp/binarydeb/ros-noetic-class-loader-0.5.0/src/class_loader.cpp

Also, I am still bombarded by messages like:
[ WARN] [1668431567.889206933, 259.155000000]: Illegal bounds change, was [tl: (-8.974486, -9.951265), br: (6.125514, 4.848735)], but is now [tl: (-4.336892, -2.193491), br: (5.990955, 1.509129)]. The offending layer is explore_costmap/polygon_layer
[ WARN] [1668431569.176657100, 260.155000000]: Illegal bounds change, was [tl: (-8.974486, -9.951265), br: (6.125514, 4.848735)], but is now [tl: (-4.336892, -2.193491), br: (5.990955, 1.509129)]. The offending layer is explore_costmap/polygon_layer
[ WARN] [1668431570.422504379, 261.155000000]: Illegal bounds change, was [tl: (-8.974486, -9.951265), br: (6.125514, 4.848735)], but is now [tl: (-4.336892, -2.193491), br: (5.990955, 1.509129)]. The offending layer is explore_costmap/polygon_layer
[ WARN] [1668431571.724391445, 262.156000000]: Illegal bounds change, was [tl: (-8.974486, -9.951265), br: (6.125514, 4.848735)], but is now [tl: (-4.336892, -2.193491), br: (5.990955, 1.509129)]. The offending layer is explore_costmap/polygon_layer