paulbovbel/frontier_exploration #54: memory leak melodic release

scifiswapnil, 2020-06-11:
[image]

hi, I was able to successfully build the package for melodic ROS. I was trying to run the basic implementation on Stage simulator when I am getting the above error. Possibly a memory leak or maybe something I am doing wrong. Is there any documentation for melodic-devel or tutorials of any sort that might help. 
Thanks.

paulbovbel, 2020-08-11:
It's not a memory leak, but an error on shutdown in terms of destruction order.

Gavriel-CTO, 2022-11-14:
I get this error as follows (Noetic - Package built from source):

The initial exploration seems to be working but ends early at some point - depends on pre-determined points AFAIK... and I get this:

[ INFO] [1668333655.747359764, 1420.770000000]: Goal reached
[ERROR] [1668333655.747967449, 1420.770000000]: BUG: Got a second transition to DONE

Then if I trying a new polygon:

[ INFO] [1668417507.443129845, 140.190000000]: Sending goal
[ INFO] [1668417507.475213758, 140.210000000]: Updating polygon
[ INFO] [1668417507.475353658, 140.210000000]: Requesting a goal
Warning: class_loader.ClassLoader: SEVERE WARNING!!!
Attempting to unload (MY_PATH)/husky_ws/devel/lib//libexample_plugin.so
while objects created by this library still exist in the heap!
You should delete your objects before destroying the ClassLoader. The library will NOT be unloaded.
at line 119 in /tmp/binarydeb/ros-noetic-class-loader-0.5.0/src/class_loader.cpp

Also, I am still bombarded by messages like:
[ WARN] [1668431567.889206933, 259.155000000]: Illegal bounds change, was [tl: (-8.974486, -9.951265), br: (6.125514, 4.848735)], but is now [tl: (-4.336892, -2.193491), br: (5.990955, 1.509129)]. The offending layer is explore_costmap/polygon_layer
[ WARN] [1668431569.176657100, 260.155000000]: Illegal bounds change, was [tl: (-8.974486, -9.951265), br: (6.125514, 4.848735)], but is now [tl: (-4.336892, -2.193491), br: (5.990955, 1.509129)]. The offending layer is explore_costmap/polygon_layer
[ WARN] [1668431570.422504379, 261.155000000]: Illegal bounds change, was [tl: (-8.974486, -9.951265), br: (6.125514, 4.848735)], but is now [tl: (-4.336892, -2.193491), br: (5.990955, 1.509129)]. The offending layer is explore_costmap/polygon_layer
[ WARN] [1668431571.724391445, 262.156000000]: Illegal bounds change, was [tl: (-8.974486, -9.951265), br: (6.125514, 4.848735)], but is now [tl: (-4.336892, -2.193491), br: (5.990955, 1.509129)]. The offending layer is explore_costmap/polygon_layer