paulbovbel/frontier_exploration #57: Not moving after draw marker with melodic

kristfrizh, 2020-08-11:
Hello guys
I'm quite new in ros and encountered some problem when I try frontier_exploration. 
I got the error after finish the drawing like this :
Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
         at line 122 in /tmp/binarydeb/ros-melodic-class-loader-0.4.1/src/class_loader.cpp

I have also tried https://github.com/paulbovbel/frontier_exploration/issues/47 this solution but still not move

I am using Jackal_simulation to do this.
thank you

kristfrizh, 2020-08-11:
[image]
[image]

here is the error and the rqt_graph

naseeef, 2020-09-24:
hi kristfrizh, I've got the same error. did you get the answer?

kristfrizh, 2020-09-24:
oh hii
still like that

I think it still not compatible with melodic version
regards

Gavriel-CTO, 2022-11-14:
I get this error as follows (Noetic - Package built from source):

The initial exploration seems to be working but ends early at some point - depends on pre-determined points AFAIK... and I get this:

[ INFO] [1668333655.747359764, 1420.770000000]: Goal reached
[ERROR] [1668333655.747967449, 1420.770000000]: BUG: Got a second transition to DONE

Then if I trying a new polygon:

[ INFO] [1668417507.443129845, 140.190000000]: Sending goal
[ INFO] [1668417507.475213758, 140.210000000]: Updating polygon
[ INFO] [1668417507.475353658, 140.210000000]: Requesting a goal
Warning: class_loader.ClassLoader: SEVERE WARNING!!!
Attempting to unload (MY_PATH)/husky_ws/devel/lib//libexample_plugin.so
while objects created by this library still exist in the heap!
You should delete your objects before destroying the ClassLoader. The library will NOT be unloaded.
         at line 119 in /tmp/binarydeb/ros-noetic-class-loader-0.5.0/src/class_loader.cpp

Also, I am still bombarded by messages like:
[ WARN] [1668431567.889206933, 259.155000000]: Illegal bounds change, was [tl: (-8.974486, -9.951265), br: (6.125514, 4.848735)], but is now [tl: (-4.336892, -2.193491), br: (5.990955, 1.509129)]. The offending layer is explore_costmap/polygon_layer
[ WARN] [1668431569.176657100, 260.155000000]: Illegal bounds change, was [tl: (-8.974486, -9.951265), br: (6.125514, 4.848735)], but is now [tl: (-4.336892, -2.193491), br: (5.990955, 1.509129)]. The offending layer is explore_costmap/polygon_layer
[ WARN] [1668431570.422504379, 261.155000000]: Illegal bounds change, was [tl: (-8.974486, -9.951265), br: (6.125514, 4.848735)], but is now [tl: (-4.336892, -2.193491), br: (5.990955, 1.509129)]. The offending layer is explore_costmap/polygon_layer
[ WARN] [1668431571.724391445, 262.156000000]: Illegal bounds change, was [tl: (-8.974486, -9.951265), br: (6.125514, 4.848735)], but is now [tl: (-4.336892, -2.193491), br: (5.990955, 1.509129)]. The offending layer is explore_costmap/polygon_layer