paulbovbel/frontier_exploration #58: Catkin_Make failed in ros kinetic ubuntu 16.04

Aronlin4458, 2020-09-19:
Hi,

When I try to Catkin_make it in my workspace. The following showed up.

/home/aron/catkin_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp: In member function ‘bool exploration_server::ExplorationServer::inBoundary()’:
/home/aron/catkin_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:144:35: error: no matching function for call to ‘costmap_2d::Costmap2DROS::getRobotPose(geometry_msgs::PoseStamped&)’
   costmap_ros_->getRobotPose(pose1);
                                   ^
In file included from /home/aron/catkin_ws/src/frontier_exploration/exploration_server/include/exploration_server/base_plugin.h:7:0,
                 from /home/aron/catkin_ws/src/frontier_exploration/exploration_server/include/exploration_server/exploration_server.h:4,
                 from /home/aron/catkin_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:1:
/opt/ros/kinetic/include/costmap_2d/costmap_2d_ros.h:123:8: note: candidate: bool costmap_2d::Costmap2DROS::getRobotPose(tf::Stamped<tf::Transform>&) const
   bool getRobotPose(tf::Stamped<tf::Pose>& global_pose) const;
        ^
/opt/ros/kinetic/include/costmap_2d/costmap_2d_ros.h:123:8: note:   no known conversion for argument 1 from ‘geometry_msgs::PoseStamped {aka geometry_msgs::PoseStamped_<std::allocator<void> >}’ to ‘tf::Stamped<tf::Transform>&’
/home/aron/catkin_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp: In member function ‘void exploration_server::ExplorationServer::requestAndSendGoal()’:
/home/aron/catkin_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:209:42: error: no matching function for call to ‘costmap_2d::Costmap2DROS::getRobotPose(geometry_msgs::PoseStamped&)’
   costmap_ros_->getRobotPose(current_pose);
                                          ^
In file included from /home/aron/catkin_ws/src/frontier_exploration/exploration_server/include/exploration_server/base_plugin.h:7:0,
                 from /home/aron/catkin_ws/src/frontier_exploration/exploration_server/include/exploration_server/exploration_server.h:4,
                 from /home/aron/catkin_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:1:
/opt/ros/kinetic/include/costmap_2d/costmap_2d_ros.h:123:8: note: candidate: bool costmap_2d::Costmap2DROS::getRobotPose(tf::Stamped<tf::Transform>&) const
   bool getRobotPose(tf::Stamped<tf::Pose>& global_pose) const;
        ^
/opt/ros/kinetic/include/costmap_2d/costmap_2d_ros.h:123:8: note:   no known conversion for argument 1 from ‘geometry_msgs::PoseStamped {aka geometry_msgs::PoseStamped_<std::allocator<void> >}’ to ‘tf::Stamped<tf::Transform>&’
[ 88%] Built target polygon_layer
In file included from /usr/include/boost/smart_ptr/make_shared.hpp:15:0,
                 from /usr/include/boost/make_shared.hpp:15,
                 from /opt/ros/kinetic/include/ros/forwards.h:38,
                 from /opt/ros/kinetic/include/ros/common.h:37,
                 from /opt/ros/kinetic/include/ros/ros.h:43,
                 from /opt/ros/kinetic/include/tf2_ros/buffer.h:38,
                 from /opt/ros/kinetic/include/tf/tf.h:48,
                 from /opt/ros/kinetic/include/costmap_2d/layer.h:43,
                 from /opt/ros/kinetic/include/costmap_2d/layered_costmap.h:42,
                 from /opt/ros/kinetic/include/costmap_2d/costmap_2d_ros.h:41,
                 from /home/aron/catkin_ws/src/frontier_exploration/exploration_server/include/exploration_server/base_plugin.h:7,
                 from /home/aron/catkin_ws/src/frontier_exploration/exploration_server/include/exploration_server/exploration_server.h:4,
                 from /home/aron/catkin_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:1:
/usr/include/boost/smart_ptr/make_shared_object.hpp: In instantiation of ‘typename boost::detail::sp_if_not_array<T>::type boost::make_shared(Args&& ...) [with T = costmap_2d::Costmap2DROS; Args = {const char (&)[16], tf2_ros::Buffer&}; typename boost::detail::sp_if_not_array<T>::type = boost::shared_ptr<costmap_2d::Costmap2DROS>]’:
/home/aron/catkin_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:33:93:   required from here
/usr/include/boost/smart_ptr/make_shared_object.hpp:254:5: error: no matching function for call to ‘costmap_2d::Costmap2DROS::Costmap2DROS(const char [16], tf2_ros::Buffer&)’
     ::new( pv ) T( boost::detail::sp_forward<Args>( args )... );
     ^
In file included from /home/aron/catkin_ws/src/frontier_exploration/exploration_server/include/exploration_server/base_plugin.h:7:0,
                 from /home/aron/catkin_ws/src/frontier_exploration/exploration_server/include/exploration_server/exploration_server.h:4,
                 from /home/aron/catkin_ws/src/frontier_exploration/exploration_server/src/exploration_server.cpp:1:
/opt/ros/kinetic/include/costmap_2d/costmap_2d_ros.h:80:3: note: candidate: costmap_2d::Costmap2DROS::Costmap2DROS(std::__cxx11::string, tf::TransformListener&)
   Costmap2DROS(std::string name, tf::TransformListener& tf);
   ^
/opt/ros/kinetic/include/costmap_2d/costmap_2d_ros.h:80:3: note:   no known conversion for argument 2 from ‘tf2_ros::Buffer’ to ‘tf::TransformListener&’
[ 90%] Linking CXX shared library /home/aron/catkin_ws/devel/lib/libexample_plugin.so
frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/build.make:62: recipe for target 'frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/src/exploration_server.cpp.o' failed
make[2]: *** [frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/src/exploration_server.cpp.o] Error 1
CMakeFiles/Makefile2:4955: recipe for target 'frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/all' failed
make[1]: *** [frontier_exploration/exploration_server/CMakeFiles/exploration_server.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 90%] Built target example_plugin
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

xiaotingbiggirl, 2020-10-15:
hi, I met a same error. Have you made it out?

numb0824, 2020-10-23:
hi,i have the same problem,could someone solve it?

numb0824, 2020-11-05:
well，i have solved this problem.use "catkin_make_isolated" instead of "catkin_make"

ZuyuanZhu, 2021-01-06:
> well，i have solved this problem.use "catkin_make_isolated" instead of "catkin_make"

unfortunately, this still does not work for me